beremiz/beremiz #11: Win10 cannot install posix-spwan

bruce202202, 2022-11-18:
How windows10 get the `spwan.h` files, On ubuntu: `apt-get install  build-essential ` can get this file.
perhaps `posix-spwan.whl`  can avoid such problem, but the relevant files are not found

```
building '_posix_spawn_cffi_fd40e3fbx1c29613d' extension
C:\Users\renxiaojiang\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\Bin\amd64\cl.exe /c /nologo /Ox /MD /W3 /GS- /DNDEBUG -IE:\work\Beremiz2\base\include -IE:\work\Beremiz2\base\PC 
`/Tcposix_spawn\__pycache__\_posix_spawn_cffi_fd40e3fbx1c29613d.c /Fobuild\temp.win-amd64-2.7\Release\posix_spawn\__pycache__\_posix_spawn_cffi_fd40e3fbx1c29613d.obj
_posix_spawn_cffi_fd40e3fbx1c29613d.c
posix_spawn\__pycache__\_posix_spawn_cffi_fd40e3fbx1c29613d.c(266) : fatal error C1083: Cannot open include file: 'spawn.h': No such file or directory
```

bruce202202, 2022-11-19:
No workaround found，Just use the `subprocess` that comes with pyhon.

etisserant, 2022-11-19:
If I remember correctly, when posix_spawn is not available, runtime uses normal fork() based subpreocess module. Posix_spawn is necessary only on some linux based RTOS, where virtual memory management can't work properly (no page fault allowed) and then makes problem with fork() in case of low memory resources. You should not need posix_spawn on windows at all.

bruce202202, 2022-11-22:
> If I remember correctly, when posix_spawn is not available, runtime uses normal fork() based subpreocess module. Posix_spawn is necessary only on some linux based RTOS, where virtual memory [...]

yes,I thought posix_spawn performance was better than subpocess

closed 2022-11-22 by bruce202202