beremiz/beremiz #12: While adding CANopen node: "No module named nodelist"

skibon02, 2022-12-03:
Hello. I am working on a runtime environment for PLC and I want to add CanOpen support. As far as I understand, I can either create my custom configuration node and specify locations that i need, or use "CANopen support" node that comes with Berimiz. But once I am trying to add it, i am getting exception 
```
canfestival/canfestival.py", line 47,
   ImportError: No module named nodelist
```
I cannot find or install such library with pip, also there are no uses of this library in the other places of project. Does anyone know what is it / where can i get it?

vitw, 2023-02-06:
Hi. 
I encountered the same problem and solved it by changing name of the canfestival directory from "canfestival" to "CanFestival-3".  
Alternatively you can change what is added to the path in canfestival/canfestival.py line 43. Change path from "CanFestival-3" to "canfestival" and everything should be working.
Also check out my pull request, if you want to get it working with wxPython4: https://github.com/beremiz/canfestival/pull/1

skibon02, 2023-02-06:
@vitw, Hi
Thank you!  I will test it later, but your investigation and PR gonna be really helpful.

closed 2023-08-08 by etisserant