beremiz/beremiz #13: EtherCAT Configurator does not run properly and many errors are reported

bruce202202, 2022-12-13:
EtherCAT Configurator does not run properly and many errors are reported, I tried to fix it, but those issues was too complicated ！
Does anyone have code that runs successfully?

skibon02, 2022-12-24:
this part of the project was probably abandoned and i guess porting to wxPython4 and writing tests are most important tasks of the project now. 
I also saw some strange errors in modbus/ettercat implementation like missing new line or semicolon and made a decision that this code was working in previous versions. 
As an alternative you can develop your own extension for beremiz and add some custom configuration nodes and locations

etisserant, 2023-08-08:
I confirm: EtherCAT support is not maintained since years, and is broken. Any contribution is welcome.

closed 2023-08-08 by etisserant