beremiz/beremiz #14: Beremiz  chrashes when connectinc with PLC

Pelikan-71-57-1, 2022-12-18:
Hello
when I try to connect to the PLC the editor crashes, but the Runtime Starts
I can replicate the Error on Tow Devices
Sadly I cannot post the Konsole output

 

I use the Snapcraft version
OS:
Device 1: Garuda Linux (Arch base)
Device 2: KDE NEON (Ubuntu base)

Pelikan-71-57-1, 2022-12-24:
[image]

Today I had some time and recorded the error message I hope this can help to fix the error

and merry Christmas

skibon02, 2022-12-24:
I am not sure but you can try current version from wxPython4 branch. At least it is working fine for me.
You also need python2 to run it

Pelikan-71-57-1, 2022-12-26:
I just tried it but the installation of wxPython fails
after returning this:  (half an hour later I get a Long and Red error Massage ) I will post it leater

> 
> 
> python2 -m pip install  wxPython==4.1.1
> DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
> Defaulting to user installation because normal site-packages is not writeable
> Collecting wxPython==4.1.1
>   Using cached wxPython-4.1.1.tar.gz (66.0 MB)
> Requirement already satisfied: pillow in ./.local/lib/python2.7/site-packages (from wxPython==4.1.1) (6.2.2)
> Requirement already satisfied: six in ./.local/lib/python2.7/site-packages (from wxPython==4.1.1) (1.16.0)
> Requirement already satisfied: numpy<1.17 in ./.local/lib/python2.7/site-packages (from wxPython==4.1.1) (1.16.5)
> Building wheels for collected packages: wxPython
>   Building wheel for wxPython (setup.py) ...

Pelikan-71-57-1, 2022-12-29:
@skibon02  may I ask what Distro you use?

skibon02, 2022-12-29:
@Pelikan-71-57-1 I use Ubuntu 22, latest version
You have to install python 2 manually if you didn't because it is not supported by default

skibon02, 2022-12-29:
> .

Your log does not contain problematic messages, it is only a part of a full log

Pelikan-71-57-1, 2022-12-30:
The problem is the full log is so long that the Terminal "overflows" and some info gets lost
any Idea how to get the full log?

Pelikan-71-57-1, 2022-12-30:
so I managed to install Beremiz on an Ubuntu VM  (just the compiler )
but even this installation crashes as son as I try to start the runtime

Ubuntu shows an error log but I cannot copy it (is there a file where the logs are stored)

Pelikan-71-57-1, 2022-12-30:
so I found the reason for the crash

the URI_location in the configuration menu has to be set manually to LOCAL:// or an valid IP Address to prevent the crash 

but I this should not crash the IDE  or should at least be pre-filled

Anyway thanks for the Help

skibon02, 2022-12-30:
If your problem is solved, you can now close the issue. Anyway, it could be convenient to show prompt if address is invalid and/or prefill it with LOCAL:// default.

Pelikan-71-57-1, 2022-12-31:
I will

But some warnings instead of crashing right away would be grate

with best regrades

closed 2022-12-31 by Pelikan-71-57-1