beremiz/beremiz #15: Beremiz bitwise handling problem

pdorokhov, 2023-06-27:
Hi. Please advise me on this question - the compiler does not allow bitwise access to bitwise strings. What can this problem be related to, because the manual says that this is correct.

[image]

[image]

etisserant, 2023-08-08:
I see nowhere such syntax in IEC-61131-3. Please point where in IEC-61131-3 is this language feature defined.

closed 2023-08-08 by etisserant

pdorokhov, 2023-08-08:
Unfortunately I cannot download the description of IEC61131-3 in English from the IEC website. I have a description of the standard in Russian, and it mentions bitwise handling, but unfortunately I can't compare it with the description of the original standard.
[image]

pdorokhov, 2023-08-08:
> I see nowhere such syntax in IEC-61131-3. Please point where in IEC-61131-3 is this language feature defined.

Unfortunately I cannot download the description of IEC61131-3 in English from the IEC website. I have a description of the standard in Russian, and it mentions bitwise handling, but unfortunately I can't compare it with the description of the original standard.
[image]

etisserant, 2023-08-08:
With your input, I could find where it is :  "6.6.1.3 Partial access to ANY_BIT variables"
This is part of IEC-61131-3 3rd edition !
3rd Edition of IEC-61131 is not supported for now, only 2nd edition is.