beremiz/beremiz #51: Codebase rework discussion

cloud-rocket, 2024-11-25:
Hi,

I am highly impressed with the project, I'd like to help making it more standardized and maintainable:

The following should be a starting point for a discussion. I suggest the following changes:

Beremiz codebase will be broke into the following standard Python packages (and Github repos), each one having a separate versioning and tests:
* `beremiz-ide`
* `beremiz-runtime`
* `beremiz-hmi` (I haven't touched the HMI part, so I don't know anything about it)
* `erpc-interface` or any other packages which are used by both ide and runtime.

This way each package will be maintainable regardless of all others.
All packages will support all the required platforms.
IDE will be released as Python package and as a bundled release for each platform.

etisserant, 2024-11-25:
Hi Meir,

Thanks for your interest in the project.

I agree that re-organizing source code to separate main components is needed. Though, I'm not sure there is any benefit in separating them into different repositories. It might actually become a versioning mess.

Couldn't this be multiple directories inside one single repository, as for example in Linux ?

Also, keep in mind that both existing tests and beremiz-public-dist will have to be updated to reflect those changes in any case.

etisserant, 2024-11-25:
If a monolithic repository is really problematic, then we could still use git submodules, but I don't think it is  a good idea. SetupTools and other distribution systems do not require individual repository per package, you can point to a python package in a sub-directory of any repository. Is there any reason why you would need separate repositories for each packages ?

cloud-rocket, 2024-11-25:
Versioning of a separate packages can be managed as the following:

- Major update - breaking change that requires to update dependencies also
- Minor - does not require to update dependencies 
- Patch - does not require to recompile PLC (just an idea)

IDE has Runtime as a dependency (to run debugging sessions locally)

The benefits of separate packages are the following:

- Separating IDE (GUI) from execution modules is always beneficiary - you can use another IDE with the same execution packages
- A bugfix of a runtime package is not related to updating GUI (usually)
- GUI always has many updates and bugfixes - this will not affect Runtime
- Runtime (and all other packages maybe except IDE) will have proper testing and coverage (you can't do coverage with GUI)
- Runtime security is far more important than IDE security - scans will be done properly. And less code is always a better security.
- You can have separate more permissive licensing on Runtime
- You can support multiple platforms under Runtime
- You can have optional dependencies which you don't have to install
- THere is a lot more....

Other considerations:
- I think submodules is always a bad idea. I never had good experience with it
- Monolithic package is nice when you run it all together. In this case you have different components running on different machines.

What are the benefits of Monolithic approach except for backward compatiblity?

etisserant, 2024-11-25:
I totally agree on your main intent : we should formalize distinguishable python packages, and publish them on Pypi.

Nevertheless none of your points manage to persuade me that there is any benefit in hosting those packages on individual repositories. Every point equally works with separate repositories or sub-directories, each with pros and cons that can be solved with either branching and tagging policies and distribution hook/action scripts.

Of course monolithic repository helps regarding backward compatibility. I don't want to annoy users that rely on this source layout in their build system for no good reason. Codebase rework will already bring enough changes that they will have to adapt to. Let's be fair, some of them are still using python2 and [old hg repo](https://hg.beremiz.org/).

But backward compatibility is not the main reason. By keeping IDE, runtime and extensions in the same repo, we can ensure consistency of the project together with tests and distribution at any time. Ensuring such consistency would otherwise require [using a thin shell repository to manage your subrepositories](https://wiki.mercurial-scm.org/Subrepository?action=show&redirect=Subrepositories#Use_a_thin_shell_repository_to_manage_your_subrepositories)

Hereafter a possible layout:

| path | description | pypi name |
|---|---|---|
| `/` | top level dir contains minimum README, COPYING, etc. | |
| `/ide` | wxPython based workbench + CLI - will appear as beremiz-ide in pypi.| beremiz-ide |
| `/rt ` | runtimes | |
| `/rt/python` | python runtime | beremiz-runtime |
| `/rt/C` | C/C++ runtimes | |
| `/defs` | common definitions : eRPC + IEC type mapping | beremiz-defs |
| `/tests` | test scripts, material and tools | |
| `/dist` | packaging and distribution (actual beremiz-public-dist) | |
| `/docs` | documentation | |

cloud-rocket, 2024-11-25:
Well, we will miss all the points I mentioned by keeping a monolitic repo (i.e. I cannot provide a source code of a runtime for a security scan by customers). Also having multiple packages for a single repo is not a standard approach.

> Ensuring such consistency would otherwise require [using a thin shell repository to manage your subrepositories]

I don't think such a project needs this consistency - You should be able to change and enhance IDE without touching or changing API of the Runtime. A version of IDE should never "freeze" runtime dependency (I.e. IDE version should limit the runtime as the following - `beremiz-runtime<3.00` to support any version 2 with its bugfixes).

Can you please explain why shell repo with sub repos will be needed? Perhaps provide an example?

Can we engage more people into this discussion? (i.e. Beremiz users?)

etisserant, 2024-11-25:
> I cannot provide a source code of a runtime for a security scan by customers

I don't understand. Customer Security Scan deals directly with whole github repository ? I suppose that it rather scans python package .whl file you provide. Of course content of this archive will not be the whole beremiz source, but only the sub-directory. And even in case Customer Security Scan would look at repository directly, then where is the problem with narrowing scan to a a sub-directory ? Or what about scanning the whole repo ? Nothing to hide here.

> I don't think such a project needs this consistency

The contrary is true. All Beremiz based PLC manufacturers I support tend to carry firmware image with new IDE version and systematically update firmware of PLC before any programming. Backward compatibility at that level is error prone and quite dangerous. Any subtle inconsistency in between the IDE, toolchain and runtime can lead to really impossible to  debug problem.

> Can you please explain why shell repo with sub repos will be needed? Perhaps provide an example?

beremiz-public-dist is a good example.

> Can we engage more people into this discussion? (i.e. Beremiz users?)

I proudly represent quite a few of them here already, but if you need greater audience, then you can still try involving random folks from OpenPLC. Unfortunately I don't think they care much about good software engineering practices.

cloud-rocket, 2024-12-09:
@etisserant ,

I hope it finds you well, but I can argue with a single repo approach. 

I started reworking the runtime into a separate repo/package doing the following:
* Separating wxPython app from the rest of the code (I still haven't published the app) and introducing CLI only way to run it.
* Reworking the `i18n` stuff to work with today's standard Python `gettext`
* Removing heavy usage of global variables and code which is not a good Python practice and causes a few problems you needed to deal with. Object based approach allows adding tests later.
* Removing unused imports and arguments
* Reformatting code with `flake8`
* Adding `console_scripts` to run the CLI 
* Adding Docker packaging (TBD)

Right now CLI seems to be working properly.

The WIP code is here - https://github.com/[cloud-rocket/beremiz-runtime](https://github.com/cloud-rocket/beremiz-runtime)

Please let me know if you see a way to collaborate on this effort.
Thanks!

cloud-rocket, 2024-12-10:
Sorry, the link was broken before

etisserant, 2024-12-17:
@cloud-rocket, code must be reviewable and loosing files history is unacceptable.

"Initial commit" commit message is too poor specially when it is not initial commit.
Furthermore, you should clone wqhole Beremiz project and use git rm and git mv to re-organize while preserving file history.

All theses change have to happen step by step in as small as possible commits so that it can easily be reviewed.

closed 2025-01-12 by etisserant