beremiz/beremiz #53: Improve wikis and documentation

HBolck, 2024-12-17:
We planned to use this tool to develop another tool, but could not find clear documentation and instructions for the project structure.
I try to find some information available on https://beremiz.readthedocs.io/en/latest/, but found that some pages are blank, such as the following parts.
Hope the author improves the Wiki and documentation.Very grateful !
 
[image]

etisserant, 2024-12-17:
In case [https://beremiz.readthedocs.io](https://beremiz.readthedocs.io/) is not up to date, sources of this documentation are in `doc` directory, but you are right, documentation is still really sparse despite of almost 20y old project.

@HBolck, not only *the* author can write documentation, everybody can. If Beremiz is useful for you or your company, please consider contributing some documentation once you will have some answers to your questions. If you need some help to develop your tool you can also ask beremiz.fr .

HBolck, 2024-12-17:
ok thanks

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/17/2024 17:40 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [beremiz/beremiz] Improve wikis and documentation (Issue #53) |

In case https://beremiz.readthedocs.io is not up to date, sources of this documentation are in doc directory, but you are right, documentation is still really sparse despite of almost 20y old project.

@HBolck, not only the author can write documentation, everybody can. If Beremiz is useful for you or your company, please consider contributing some documentation once you will have some answers to your questions. If you need some help to develop your tool you can also ask beremiz.fr .

HBolck, 2024-12-18:
> In case [https://beremiz.readthedocs.io](https://beremiz.readthedocs.io/) is not up to date, sources of this documentation are in `doc` directory, but you are right, documentation is still really [...]

Are there any fixed formats or requirements for contributing documents? I'm going to take some time to sort out the project structure and refine a document.

HBolck, 2024-12-19:
[image]
I briefly combed through the directory structure and reference relationships. In the figure, the blue text block represents the package, the black text block represents the file, the solid line represents the include, and the dashed line represents the reference relationship.
@etisserant